grafana-cold-storage/opcua-datasource #5: Cannot load data source in Gragana 6.7.1

ajaywazir, 2020-03-22:
When adding opc UA datasource, I receive success pop up but the screen shows 

Unexpected token '<' Evaluating http://localhost:8800/public/plugins/grafana-opcua-datasource/module.js Loading plugins/grafana-opcua-datasource/module
Data Source Error

Please let me know how to fix

Ajay

srclosson, 2020-06-09:
@ajaywazir: Did you ever figure this out? Sorry... it's been like 2 1/2 months

ajaywazir, 2020-06-21:
Hi 
Sorry. I did not try afterwards

wuzhoubr, 2020-08-26:
I also meet this issue：Unexpected token '<' Evaluating http://127.0.0.1:8080/public/plugins/grafana-opcua-datasource/module.js Loading plugins/grafana-opcua-datasource/module
how to deal with it.

Toha1997, 2020-09-21:
I got the same error because I forgot to build from the source files in first place. Make sure to run the `yarn` and `make` commands and reload Grafana afterward. You have to install both yarn (https://classic.yarnpkg.com/en/docs/install/#windows-stable) and make (http://gnuwin32.sourceforge.net/packages/make.htm) beforehand.

mimimath14, 2020-12-29:
Hi team,
I’m neewbe here and I never use a yarn and a make with windows 10 tool to resolve this issue:
Unexpected token '<' Evaluating http://127.0.0.1:8080/public/plugins/grafana-opcua-datasource/module.js Loading 

someone could explain me how to do it?

many thanks by advance

closed 2021-09-28 by alydersen