grafana-cold-storage/opcua-datasource #9: Making the plugin more widely known - list it in Grafana plugin directory

xkilian, 2020-06-03:
Is it possible to validate that the plugin works with Grafana 7 and put it in the official Grafana list of plugins. A lot of people are looking for this plugin, and don't know about it and can't find it because Google doesn't list it in searches.
I know this will generate feature requests and what not, but maybe someone else will step up and keep improving it.

This plugin + prometheus modbus_exporter + prometheus snmp_exporter + prometheus + flowcharting + worldmap + regular grafana panels + divpanel = modern industrial/IOT open source historian UI.

srclosson, 2020-06-09:
Hi @xkilian I do have a branch for these improvements, that will hopefully include streaming eventually.

Take a look at https://github.com/srclosson/grafana-opcua-datasource/tree/latest-protobuf-and-dataframes

The only reason it's not merged yet is because the read processed does not work (read raw does)...

alydersen, 2021-09-28:
@xkilian Just wanted to let you know that @srclosson has done a great job and this plugin is actually in the official plugin list now :-)

closed 2021-09-28 by alydersen