grafana-cold-storage/opcua-datasource #12: install question

stigersmile, 2020-07-17:
Hi Thank you for your toolkit,

But after I run  follow your installation guide and restart my grafana
`
    yarn install to install dependencies
    yarn build | yarn dev to build the plugin
    make build to build the backend component
` 
I didn't see any new datasouce plugin in? (will it plugin automatically)
my grafana version is V7.1.0

dvdjda, 2020-07-28:
I got the same, after all no plugin appeared in grafana

MrLight, 2020-08-14:
You have to add an exception into the configuration to allow loading unsigned plugins.

`allow_loading_unsigned_plugins = grafana-opcua-datasource`

After that the plugin shows up in the list of datasources.

alydersen, 2021-06-07:
I'm closing this due to lack of response. Thanks to @MrLight for answering.

closed 2021-06-07 by alydersen