grafana-cold-storage/opcua-datasource #14: Failed to start plugin - gpx_opcua_windows_amd64.exe : file does not exist

kevinterface, 2020-08-15:
While trying to get the plugin up and running i encounter the following error entry in the grafana.log (located in ..\grafana\data\log)

- t=2020-08-15T14:56:29+0200 lvl=warn msg="Running an unsigned backend plugin" logger=plugins pluginID=grafana-opcua-datasource pluginDir="C:\\Program Files\\GrafanaLabs\\grafana\\data\\plugins\\grafana-opcua-datasource\\dist"
- t=2020-08-15T14:56:29+0200 lvl=info msg="Registering plugin" logger=plugins name="OPC UA (Unified Architecture)"
- t=2020-08-15T14:56:30+0200 lvl=info msg="Registering plugin" logger=plugins name=Redis
- t=2020-08-15T14:56:30+0200 lvl=eror msg="**Failed to start plugin**" logger=plugins.backend pluginId=grafana-opcua-datasource error="exec: \"C:\\\\Program Files\\\\GrafanaLabs\\\\grafana\\\\data\\\\plugins\\\\grafana-opcua-datasource\\\\dist/**gpx_opcua_windows_amd64.exe\": file does not exist"**
- t=2020-08-15T14:56:30+0200 lvl=info msg="HTTP Server Listen" logger=http.server address=[::]:3000 protocol=http subUrl= socket=
- t=2020-08-15T14:56:44+0200 lvl=eror msg="Plugin health check failed" logger=context userId=1 orgId=1 uname=admin error="Failed to check plugin health: Health check failed" remote_addr=[::1]
- t=2020-08-15T14:56:44+0200 lvl=eror msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=GET path=/api/datasources/2/health status=500 remote_addr=[::1] time_ms=27 size=40 referer=http://localhost:3000/datasources/edit/2/

I see a reference in the following file ..\grafana\data\plugins\grafana-opcua-datasource\pkg\dotnet\plugin-dotnet\.win.build.csproj but also a comment regarding the issue of the missing file.

`  <PropertyGroup>
    
    <Protobuf_NoWarnMissingExpected>true</Protobuf_NoWarnMissingExpected>
    <AssemblyName>gpx_opcua_darwin_amd64</AssemblyName>
  </PropertyGroup>`

Are you also seeing this log entry in your grafana.log? At the moment I cannot connect to my OPC UA Server, when using UAExpert Client the server is available and I can read data from it.

MrLight, 2020-08-17:
for me the usage of the following command helps to build not a darwin file but a windows exec file.
I assume that you have a gpx_opcua_darwin_amd64.exe file in your dist folder.

`dotnet publish ./pkg/dotnet/plugin-dotnet/.win.build.csproj -r win-x64 -o ./dist --self-contained true`

srclosson, 2020-08-26:
Oh, yes. That should do it (the command above)

Toha1997, 2020-09-21:
Exactly what i needed! 👍

closed 2021-06-07 by alydersen