grafana-cold-storage/opcua-datasource #15: "Plugin health check failed" after Grafana restart

Toha1997, 2020-09-21:
Hello,

I successfully installed the plugin today. I got no errors in the grafana.log file and try to configure the OPC-UA connection. I restarted the Grafana server because I could not connect to my OPC UA Server (due to wrong Opc UA settings). After the restart,  Grafana outputs "Plugin health check failed" every time I press "Save & Test". Here is corresponding the grafana.log:

`t=2020-09-21T15:04:59+0200 lvl=info msg="Starting Grafana" logger=server version=7.1.5 commit=9893b8c53d branch=HEAD compiled=2020-08-25T10:27:17+0200
t=2020-09-21T15:04:59+0200 lvl=info msg="Config loaded from" logger=settings file="C:\\Program Files\\GrafanaLabs\\grafana/conf/defaults.ini"
t=2020-09-21T15:04:59+0200 lvl=info msg="Config loaded from" logger=settings file="C:\\Program Files\\GrafanaLabs\\grafana\\conf\\custom.ini"
t=2020-09-21T15:04:59+0200 lvl=info msg="Path Home" logger=settings path="C:\\Program Files\\GrafanaLabs\\grafana"
t=2020-09-21T15:04:59+0200 lvl=info msg="Path Data" logger=settings path="C:\\Program Files\\GrafanaLabs\\grafana\\data"
t=2020-09-21T15:04:59+0200 lvl=info msg="Path Logs" logger=settings path="C:\\Program Files\\GrafanaLabs\\grafana\\data\\log"
t=2020-09-21T15:04:59+0200 lvl=info msg="Path Plugins" logger=settings path="C:\\Program Files\\GrafanaLabs\\grafana\\data\\plugins"
t=2020-09-21T15:04:59+0200 lvl=info msg="Path Provisioning" logger=settings path="C:\\Program Files\\GrafanaLabs\\grafana\\conf\\provisioning"
t=2020-09-21T15:04:59+0200 lvl=info msg="App mode production" logger=settings
t=2020-09-21T15:04:59+0200 lvl=info msg="Connecting to DB" logger=sqlstore dbtype=sqlite3
t=2020-09-21T15:04:59+0200 lvl=info msg="Starting DB migration" logger=migrator
t=2020-09-21T15:04:59+0200 lvl=info msg="Starting plugin search" logger=plugins
t=2020-09-21T15:04:59+0200 lvl=info msg="Registering plugin" logger=plugins name="Direct Input"
t=2020-09-21T15:04:59+0200 lvl=warn msg="Running an unsigned backend plugin" logger=plugins pluginID=grafana-opcua-datasource pluginDir="C:\\Program Files\\GrafanaLabs\\grafana\\data\\plugins\\grafana-opcua-datasource-master\\dist"
t=2020-09-21T15:04:59+0200 lvl=info msg="Registering plugin" logger=plugins name="OPC UA (Unified Architecture)"
t=2020-09-21T15:04:59+0200 lvl=info msg="HTTP Server Listen" logger=http.server address=[::]:3000 protocol=http subUrl= socket=
t=2020-09-21T15:05:12+0200 lvl=eror msg="Plugin health check failed" logger=context userId=1 orgId=1 uname=admin error="Failed to check plugin health: Health check failed" remote_addr=[::1]
t=2020-09-21T15:05:12+0200 lvl=eror msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=GET path=/api/datasources/10/health status=500 remote_addr=[::1] time_ms=496 size=40 referer=http://localhost:3000/datasources/edit/10/`

If you need any additional information just contact me.
Many thanks in advance and have a nice day!

Tom

wuzhoubr, 2020-09-23:
Thanks for your suggestion about another issue before，today I yarn install and yarn build|yarn dev successfully.
But I make build failed ，it shows “make（e=2，The specified file could not be found ）”，then I try to connect to OPCUA server，I also meet the same error"Pulgin health check failed",I test it in win 7(64bit) and win 10, all cannot work.

Toha1997, 2020-09-23:
Without a successful "make build" the backend component is not compiled. So that is why you get the **Plugin health check failed**.  Unfortunately, I don't know why your "make" fails. Maybe the developers will comment as well 😄.

Toha1997, 2020-09-24:
I reinstalled the plugin today and now everything works as intended 👍

closed 2020-09-24 by Toha1997