grafana-cold-storage/opcua-datasource #17: Build failed

silence3322, 2021-02-03:
>PS E:\nodeJsLearing\grafana-opcua-datasource> yarn build                                                                                                                                                  yarn run v1.22.10
$ grafana-toolkit plugin:build && make
√ Preparing
√ Linting
 PASS  src/module.test.ts
  placeholder test
    √ should return true (2ms)

Test Suites: 1 passed, 1 total
Tests:       1 passed, 1 total
Snapshots:   0 total
Time:        2.54s
Ran all test suites with tests matching "".
√ Running tests
\ Compiling...  Starting type checking service...
  Using 1 worker with 2048MB memory limit
/ Compiling...  ERROR in E:/nodeJsLearing/grafana-opcua-datasource/src/components/TreeEditor.tsx(18,31):
  TS2344: Type 'DataSource' does not satisfy the constraint 'DataSourceApi<OpcUaQuery, OpcUaDataSourceOptions>'.
  ERROR in E:/nodeJsLearing/grafana-opcua-datasource/src/module.ts(7,44):
  TS2344: Type 'DataSource' does not satisfy the constraint 'DataSourceApi<OpcUaQuery, OpcUaDataSourceOptions>'.
    The types returned by 'query(...)' are incompatible between these types.
      Type 'Observable<DataQueryResponse>' is not assignable to type 'Promise<DataQueryResponse> | Observable<DataQueryResponse>'.
        Type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/Observable").Observable<import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/types/datasource").DataQueryResponse>' is not assignable to type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Observable").Observable<import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/types/datasource").DataQueryResponse>'.
  ERROR in E:/nodeJsLearing/grafana-opcua-datasource/src/module.ts(8,20):
  TS2345: Argument of type 'typeof ConfigEditor' is not assignable to parameter of type 'ComponentType<DataSourcePluginOptionsEditorProps<OpcUaDataSourceOptions, unknown>>'.
    Type 'typeof ConfigEditor' is not assignable to type 'ComponentClass<DataSourcePluginOptionsEditorProps<OpcUaDataSourceOptions, unknown>, any>'.
      Types of parameters 'props' and 'props' are incompatible.
        Type 'DataSourcePluginOptionsEditorProps<OpcUaDataSourceOptions, unknown>' is not assignable to type 'Props | Readonly<Props>'.
          Type 'DataSourcePluginOptionsEditorProps<OpcUaDataSourceOptions, unknown>' is not assignable to type 'Readonly<Props>'.
            Types of property 'options' are incompatible.
              Type 'DataSourceSettings<OpcUaDataSourceOptions, unknown>' is not assignable to type 'DataSourceSettings<OpcUaDataSourceOptions, {}>'.
                Type 'unknown' is not assignable to type '{}'.
  ERROR in E:/nodeJsLearing/grafana-opcua-datasource/src/QueryEditor.tsx(21,31):
  TS2344: Type 'DataSource' does not satisfy the constraint 'DataSourceApi<OpcUaQuery, OpcUaDataSourceOptions>'.
    The types returned by 'query(...)' are incompatible between these types.
      Type 'Observable<DataQueryResponse>' is not assignable to type 'Promise<DataQueryResponse> | Observable<DataQueryResponse>'.
        Type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/Observable").Observable<import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/types/datasource").DataQueryResponse>' is not assignable to type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Observable").Observable<import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/types/datasource").DataQueryResponse>'.
          The types of 'source.operator.call' are incompatible between these types.
            Type '(subscriber: import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/Subscriber").Subscriber<any>, source: any) => import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/types").TeardownLogic' is not assignable to type '(subscriber: import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Subscriber").Subscriber<any>, source: any) => import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/types").TeardownLogic'.
              Types of parameters 'subscriber' and 'subscriber' are incompatible.
                Type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Subscriber").Subscriber<any>' is not assignable to type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/Subscriber").Subscriber<any>'.
                  Property 'isStopped' is protected but type 'Subscriber<T>' is not a class derived from 'Subscriber<T>'.

   Hash: b0e80cf97d0082983700
  Version: webpack 4.41.5
  Time: 17532ms
  Built at: 2021-02-03 10:02:13 ├F10: AM┤
                  Asset       Size  Chunks                          Chunk Names
           CHANGELOG.md  193 bytes          [emitted]
                LICENSE   11.3 KiB          [emitted]
              README.md   2.42 KiB          [emitted]
     img/dashboard1.png    189 KiB          [emitted]
     img/dashboard2.png    907 KiB          [emitted]        [big]
           img/logo.svg  876 bytes          [emitted]
       img/opc_logo.png   63.1 KiB          [emitted]
              module.js    152 KiB       0  [emitted]               module
  module.js.LICENSE.txt   1.31 KiB          [emitted]
          module.js.map    677 KiB       0  [emitted] [dev]         module
            plugin.json   1.02 KiB          [emitted]
  Entrypoint module = module.js module.js.map
   [0] external "react" 42 bytes {0} [built]
   [1] external "@grafana/ui" 42 bytes {0} [built]
  [11] external "emotion" 42 bytes {0} [built]
  [13] external "@grafana/data" 42 bytes {0} [built]
  [14] external "@grafana/runtime" 42 bytes {0} [built]
  [21] ../node_modules/@material-ui/core/esm/TableBody/index.js 38 bytes [built]
  [22] ../node_modules/@material-ui/core/esm/TableCell/index.js 38 bytes [built]
  [23] ../node_modules/@material-ui/core/esm/TableRow/index.js 37 bytes [built]
  [24] ../node_modules/@material-ui/core/esm/Table/index.js 34 bytes [built]
  [25] ../node_modules/@material-ui/core/esm/index.js 10.3 KiB [built]
[... 42 lines cut ...]
              Type 'DataSourceSettings<OpcUaDataSourceOptions, unknown>' is not assignable to type 'DataSourceSettings<OpcUaDataSourceOptions, {}>'.
                Type 'unknown' is not assignable to type '{}'.

  ERROR in E:/nodeJsLearing/grafana-opcua-datasource/src/QueryEditor.tsx
  ERROR in E:/nodeJsLearing/grafana-opcua-datasource/src/QueryEditor.tsx(21,31):
  TS2344: Type 'DataSource' does not satisfy the constraint 'DataSourceApi<OpcUaQuery, OpcUaDataSourceOptions>'.
    The types returned by 'query(...)' are incompatible between these types.
      Type 'Observable<DataQueryResponse>' is not assignable to type 'Promise<DataQueryResponse> | Observable<DataQueryResponse>'.
        Type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/Observable").Observable<import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/types/datasource").DataQueryResponse>' is not assignable to type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Observable").Observable<import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/types/datasource").DataQueryResponse>'.
          The types of 'source.operator.call' are incompatible between these types.
            Type '(subscriber: import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/Subscriber").Subscriber<any>, source: any) => import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/rxjs/internal/types").TeardownLogic' is not assignable to type '(subscriber: import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Subscriber").Subscriber<any>, source: any) => import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/types").TeardownLogic'.
              Types of parameters 'subscriber' and 'subscriber' are incompatible.
                Type 'import("E:/nodeJsLearing/grafana-opcua-datasource/node_modules/@grafana/data/node_modules/rxjs/internal/Subscriber").Subscriber<any>' is not assignable to type 'import("E:/nodeJsLearing/grafana-opcua-datasource/n [... cut ...]
      at Object.throw (E:\nodeJsLearing\grafana-opcua-datasource\node_modules\@grafana\toolkit\node_modules\tslib\tslib.js:121:57)
      at rejected (E:\nodeJsLearing\grafana-opcua-datasource\node_modules\@grafana\toolkit\node_modules\tslib\tslib.js:112:69)
      at runMicrotasks (<anonymous>)
      at processTicksAndRejections (internal/process/task_queues.js:97:5)
× Build failed
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

silence3322, 2021-02-03:
My node version: v12.18.2 on windows.

colsieb, 2021-03-01:
I wonder if I can bump this thread if I may?
I cannot get this to build either, with what looks like the exact same compile errors.  I have tried various permutations of the build but to no avail.
I'm running Grafana 7.2.2 on Debian9 Stretch, NodeJS v15.1, NPM7.5.3

It installs all the packages ok, albeit with few dependency and dependency warnings.  Just won't build.
Any pointers would be much appreciated! What sort of environments have others had success in?

It would be great to get this working as I currently use Grafana in our manufacturing plants, supplementing the SCADA system to give great dashboards; OPC-UA would be an awesome addition.

amir135, 2021-03-03:
I get same error.

soniakarkar, 2021-04-10:
Same error here.
Completely new to TypeScript so I have a hard time even starting to search for a hint. From quick googling it feels like it could "just" be a pbm of versions ?
Like the others, a bunch of warning about unmet dependencies at install, then same error at build.

Ubuntu 20.04
Grafana v7.5.3 (3e3cf4dc60)
nodejs v10.19.0
npm 6.14.8
yarn-1.22.10
typescript@4.2.4

<pre><font color="#C4A000">warning</font> &quot; &gt; @material-ui/core@4.11.2&quot; has unmet peer dependency &quot;react@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot; &gt; @material-ui/core@4.11.2&quot; has unmet peer dependency &quot;react-dom@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; @material-ui/styles@4.11.2&quot; has unmet peer dependency &quot;react@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; @material-ui/styles@4.11.2&quot; has unmet peer dependency &quot;react-dom@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; @material-ui/system@4.11.2&quot; has unmet peer dependency &quot;react@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; @material-ui/system@4.11.2&quot; has unmet peer dependency &quot;react-dom@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; @material-ui/utils@4.11.2&quot; has unmet peer dependency &quot;react@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; @material-ui/utils@4.11.2&quot; has unmet peer dependency &quot;react-dom@^16.8.0 || ^17.0.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; react-transition-group@4.4.1&quot; has unmet peer dependency &quot;react@&gt;=16.6.0&quot;.
<font color="#C4A000">warning</font> &quot;@material-ui/core &gt; react-transition-group@4.4.1&quot; has unmet peer dependency &quot;react-dom@&gt;=16.6.0&quot;.
<font color="#C4A000">warning</font> &quot; &gt; antd@3.26.20&quot; has unmet peer dependency &quot;react@&gt;=16.0.0&quot;.
<font color="#C4A000">warning</font> &quot; &gt; antd@3.26.20&quot; has unmet peer dependency &quot;react-dom@&gt;=16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; @ant-design/create-react-context@0.2.5&quot; has unmet peer dependency &quot;react@^0.14.0 || ^15.0.0 || ^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; @ant-design/icons-react@2.0.1&quot; has unmet peer dependency &quot;react@16.x&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-drawer@3.1.3&quot; has unmet peer dependency &quot;react@*&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-editor-mention@1.1.13&quot; has unmet peer dependency &quot;react@&gt;=15.x&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-editor-mention@1.1.13&quot; has unmet peer dependency &quot;react-dom@&gt;=15.x&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-mentions@0.4.2&quot; has unmet peer dependency &quot;react@*&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-menu@7.5.5&quot; has unmet peer dependency &quot;react@*&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-menu@7.5.5&quot; has unmet peer dependency &quot;react-dom@*&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-resize-observer@0.1.3&quot; has unmet peer dependency &quot;react@^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-resize-observer@0.1.3&quot; has unmet peer dependency &quot;react-dom@^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-switch@1.9.2&quot; has unmet peer dependency &quot;react@^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-switch@1.9.2&quot; has unmet peer dependency &quot;react-dom@^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-table@6.10.15&quot; has unmet peer dependency &quot;react@^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-table@6.10.15&quot; has unmet peer dependency &quot;react-dom@^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-tabs@9.7.0&quot; has unmet peer dependency &quot;react@&gt;=15.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-tree@2.1.4&quot; has unmet peer dependency &quot;react@*&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-tree@2.1.4&quot; has unmet peer dependency &quot;react-dom@*&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; react-lazy-load@3.1.13&quot; has unmet peer dependency &quot;react@^0.14.0 || ^15.0.0-0 || ^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; react-lazy-load@3.1.13&quot; has unmet peer dependency &quot;react-dom@^0.14.0 || ^15.0.0-0 || ^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; react-slick@0.25.2&quot; has unmet peer dependency &quot;react@^0.14.0 || ^15.0.1 || ^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; react-slick@0.25.2&quot; has unmet peer dependency &quot;react-dom@^0.14.0 || ^15.0.1 || ^16.0.0&quot;.
<font color="#C4A000">warning</font> &quot;antd &gt; rc-editor-mention &gt; draft-js@0.10.5&quot; has unmet peer dependency &quot;react@^0.14.0 || ^15.0.0-rc || ^16.0.0-rc || ^16.0.0&quot;.
[... cut ...]
    8:1  <font color="#CC0000">error</font>  &apos;@grafana/ui&apos; import is duplicated  no-duplicate-imports
  
  <u style="text-decoration-style:single">grafana-opcua-datasource/src/components/Cascader/Cascader.tsx</u>
    4:1  <font color="#CC0000">error</font>  &apos;@grafana/ui&apos; import is duplicated  no-duplicate-imports
    5:1  <font color="#CC0000">error</font>  &apos;@grafana/ui&apos; import is duplicated  no-duplicate-imports
  
  <font color="#CC0000"><b>✖ 6 problems (6 errors, 0 warnings)</b></font>
</pre>

any help, hint direction pointing would be deeply appreciated.

alydersen, 2021-06-07:
@soniakarkar  The double imports have been fixed in the latest PR and builds nicely with `yarn build`.

alydersen, 2021-09-28:
@soniakarkar This datasource is in the official plugin list now, so you don't have to build it anymore :-)

closed 2021-09-28 by alydersen