grafana-cold-storage/opcua-datasource #32: Timestamp not returning correctly.

JustinTweedie, 2021-06-23:
I have compiled the plugin on a Ubuntu machine with the latest Grafana version.
The OPC UA connects correctly but all timestamps are "1970-01-01 02:27:04".

alydersen, 2021-06-25:
Hi @JustinTweedie !

Have you tried to use the dome OPC UA server at opc.tcp://opcua.prediktor.com:4880 to see if you get the same issue with timestamps there?

JustinTweedie, 2021-06-25:
The following is returned.

1970-01-01 02:27:04
75.9
1970-01-01 02:27:04
84.4
1970-01-01 02:27:04
92.8
1970-01-01 02:27:04
101
1970-01-01 02:27:04
107
1970-01-01 02:27:04
115
1970-01-01 02:27:04
124
1970-01-01 02:27:04
132
1970-01-01 02:27:04
141
1970-01-01 02:27:04
149
1970-01-01 02:27:04
155
1970-01-01 02:27:04
163
1970-01-01 02:27:04
172
1970-01-01 02:27:04
-180
1970-01-01 02:27:04
-172
1970-01-01 02:27:04
-163
1970-01-01 02:27:04
-155
1970-01-01 02:27:04
-149
1970-01-01 02:27:04
-141
1970-01-01 02:27:04
-132
1970-01-01 02:27:04
-124
1970-01-01 02:27:04
-115
1970-01-01 02:27:04
-107
1970-01-01 02:27:04
-101
1970-01-01 02:27:04
-93
1970-01-01 02:27:04
-84

Best Regards
Justin Tweedie

*WEBSITE*
www.acusys.co.za

*QSE, Level 2 BEE Contributor*
*Justin Tweedie - Managing Director*
AcuSYS Software Solutions (Pty) Ltd
604 Broadbury Circle, Irene, Gauteng, South Africa

T: +27 (12) 667-1889
M: +27 (82) 814-9431
E: ***@***.***
 Website <http://www.acusys.co.za> |   vCard
<http://www.acusys.co.za/images/TweedieJustin.vcf> |   LinkedIn
<https://www.linkedin.com/in/justintweedie>

sleiban, 2021-06-25:
Hi, Justin!
What kind of aggregation have you specified? Raw or Interpolative is usually good choices.

[image]

Br.
Oddvar

JustinTweedie, 2021-06-25:
Using RAW.

Tried both and the result is the same.
Is there a log file somewhere?

Best Regards
Justin Tweedie

*WEBSITE*
www.acusys.co.za

*QSE, Level 2 BEE Contributor*
*Justin Tweedie - Managing Director*
AcuSYS Software Solutions (Pty) Ltd
604 Broadbury Circle, Irene, Gauteng, South Africa

T: +27 (12) 667-1889
M: +27 (82) 814-9431
E: ***@***.***
 Website <http://www.acusys.co.za> |   vCard
<http://www.acusys.co.za/images/TweedieJustin.vcf> |   LinkedIn
<https://www.linkedin.com/in/justintweedie>

sleiban, 2021-06-25:
You should find log files beneath the plugin directory.  plugins\opcua\logs

Br.
Oddvar

JustinTweedie, 2021-06-25:
Logs show nothing :(
No error / warning

Best Regards
Justin Tweedie

*WEBSITE*
www.acusys.co.za

*QSE, Level 2 BEE Contributor*
*Justin Tweedie - Managing Director*
AcuSYS Software Solutions (Pty) Ltd
604 Broadbury Circle, Irene, Gauteng, South Africa

T: +27 (12) 667-1889
M: +27 (82) 814-9431
E: ***@***.***
 Website <http://www.acusys.co.za> |   vCard
<http://www.acusys.co.za/images/TweedieJustin.vcf> |   LinkedIn
<https://www.linkedin.com/in/justintweedie>

alydersen, 2021-06-25:
Hi Justin,

I'd think this is something wrong in the build/compile, as this doesn't happen other places we've seen. Could you try to build it on another machine?

JustinTweedie, 2021-06-25:
This is on a fresh install of Ubuntu

After the Yarn and Make I copied the "dist" folder to
/var/lib/grafana/plugins/opcua

The plugin is unsigned so I had to configure grafana to allow for that.

Connection is good and data is refreshed just the datetime not updating.

Any suggestions on the build and install process?

Best Regards
Justin Tweedie

*WEBSITE*
www.acusys.co.za

*QSE, Level 2 BEE Contributor*
*Justin Tweedie - Managing Director*
AcuSYS Software Solutions (Pty) Ltd
604 Broadbury Circle, Irene, Gauteng, South Africa

T: +27 (12) 667-1889
M: +27 (82) 814-9431
E: ***@***.***
 Website <http://www.acusys.co.za> |   vCard
<http://www.acusys.co.za/images/TweedieJustin.vcf> |   LinkedIn
<https://www.linkedin.com/in/justintweedie>

alydersen, 2021-09-28:
Hello @JustinTweedie,

Could it be that https://github.com/grafana/opcua-datasource/commit/470f76fed44d3d4754bb43181ebc8f86efd4e38e have fixed your issue? It is part of the main branch now.

alydersen, 2021-09-28:
@JustinTweedie The plugin is listed in the official store now, so that should make it a lot easier for you to test it again. Thanks to @srclosson :-)

closed 2021-09-28 by alydersen