grafana-cold-storage/opcua-datasource #34: Redundancy support

alydersen, 2021-09-29:
It would be nice to be able to connect the data source plugin to a redundant set of OPC UA servers
so that Grafana retrieves data from the server node exposing the highest service level and automatically connects to an alternative node if the service level of the existing node goes below 200. (See OPC UA spec).

We (Prediktor) will look into this, but wanted to registerer the issue/feature request

ChrisHoll13, 2022-06-14:
Have you guys at Prediktor managed to get anywhere with this? I am looking at implementing a solution in the next few months and this would be really useful.

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won't receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson