grafana-cold-storage/opcua-datasource #41: Plugin Unavailable

crazycto, 2021-11-22:
Hello,

Using a fresh docker container of Grafana v8.2.5 (b57a137acd)
and the latest version available of this plugin (1.1.4) i get plugin unavailable errors.

These are some logs i get:

```
t=2021-11-22T14:38:43+0000 lvl=warn msg="plugin failed to exit gracefully" logger=plugins.backend pluginId=grafana-opcua-datasource
t=2021-11-22T14:38:45+0000 lvl=eror msg="Failed to start plugin" logger=plugins.backend pluginId=grafana-opcua-datasource error="Unrecognized remote plugin message: \n\nThis usually means that the plugin is either invalid or simply\nneeds to be recompiled to support the latest protocol."
t=2021-11-22T14:38:45+0000 lvl=info msg="Registering plugin" logger=plugins id=grafana-opcua-datasource
t=2021-11-22T14:46:41+0000 lvl=warn msg="plugin failed to exit gracefully" logger=plugins.backend pluginId=grafana-opcua-datasource
```

crazycto, 2021-11-22:
I tried installing Grafana Enterprise v8.2.5 (b57a137acd) on Windows 10 Pro and then installing the plugin, but i get the same error

from the logs:
```
t=2021-11-22T16:05:58+0100 lvl=info msg="Registering plugin" logger=plugins id=input
t=2021-11-22T16:05:58+0100 lvl=warn msg="The following files were not included in the signature" logger=plugins plugin=grafana-opcua-datasource files="[dbs/dashboardmapping.db logs/debug.log4net logs/error.log4net logs/info.log4net logs/warning.log4net]"
t=2021-11-22T16:05:58+0100 lvl=warn msg="Some plugin scanning errors were found" logger=plugins errors="plugin 'grafana-opcua-datasource' has a modified signature"
t=2021-11-22T16:05:58+0100 lvl=info msg="Live Push Gateway initialization" logger=live.push_http
t=2021-11-22T16:05:58+0100 lvl=info msg="HTTP Server Listen" logger=http.server address=[::]:3000 protocol=http subUrl= socket=
t=2021-11-22T16:05:58+0100 lvl=warn msg="Scheduling and sending of reports disabled, SMTP is not configured and enabled. Configure SMTP to enable." logger=report
t=2021-11-22T16:05:59+0100 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=GET path=/api/live/ws status=0 remote_addr=192.168.1.156 time_ms=673 size=0 referer=
t=2021-11-22T16:06:02+0100 lvl=eror msg="Could not find plugin definition for data source: grafana-opcua-datasource"
```

schall1337, 2021-11-24:
I get the same error

alydersen, 2021-11-24:
@srclosson Is this related to the work done on the last release?

crazycto, 2021-11-25:
Removing the MANIFEST file and whitelisting the plug-in in the config file solves the issue. 

Also, straight after installing the plug-in on a fresh install of grafana it works, after the first restart of the service it breaks.

srclosson, 2021-11-26:
Hmmmm.... none of the files have been modified @crazycto ? The point of the manifest is to ensure none of the files were tampered with.

schall1337, 2021-11-29:
any news how to fix this?

crazycto, 2022-01-08:
@srclosson No, nothing was changed by us on our installation, we just installed the extension from the store and then by cli on a fresh install and then this happens.

We removed the manifest since we thought that maybe there was some issues with file integrity, but it's not a fix, it's a workaround.

crazycto, 2022-01-08:
Also as you can note from the logs I shared, it seems to be in conflict with log files and a db file: 

`t=2021-11-22T16:05:58+0100 lvl=warn msg="The following files were not included in the signature" logger=plugins plugin=grafana-opcua-datasource files="[dbs/dashboardmapping.db logs/debug.log4net logs/error.log4net logs/info.log4net logs/warning.log4net]" `

Are they created after first run? Maybe that's the issue here.

sleiban, 2022-02-11:
Hei, @crazycto! A new release is out. Thank you @srclosson for making this happen! 
If you are running Linux, please note that you have to add something to the Grafana config file, see the Configuration section in the README.

crazycto, 2022-02-11:
Thanks to everyone for the support. We’ll try this asap!

closed 2022-02-11 by crazycto