grafana-cold-storage/opcua-datasource #42: Modified signature

schall1337, 2021-11-23:
Hi,
after installing the plugin and restarting Grafana I get the message "Modified signature". 
Windows 10, Grafana v8.2.5 (b57a137acd)
How can i fix this? 
Cheers

crazycto, 2021-11-24:
Yeah, troubleshooting #41 i got that as well...

alydersen, 2021-11-24:
@srclosson Is this related to the work done on the last release?

JoeH20, 2022-02-08:
Anytime I modify the grafana.ini file and restart the service the plugin is disabled for "Modified signature".

Currently running Grafana on windows 

Are there any updates on this issue?

sleiban, 2022-02-09:
@JoeH20 As a workaround untill the next release is ready (hopefully soon now), you can delete the MANIFEST file for the plugin and allow for unsigned plugin.

In Custom.ini, add the following under [plugins] section:
allow_loading_unsigned_plugins = grafana-opcua-datasource

srclosson, 2022-02-11:
This should now be fixed. Would you try the latest 1.1.5 release and respond back?

sleiban, 2022-02-15:
This is strange.
How do you install this plugin? 
Are you cleaning up any existing folders containing this plugin?
Do you download from https://grafana.com/grafana/plugins/grafana-opcua-datasource/?tab=installation ?
Then extract it to C:\Program Files\GrafanaLabs\grafana\data\plugins ?

schall1337, 2022-02-15:
For me the modified signature seems now to be resolved, though when i try to test the plugin as a datasource i get the message "plugin health check failed".

srclosson, 2022-02-17:
In that case, something else is usually going on. Have you looked at the logs @schall1337?
Another idea is to run the plugin from the command line. In c:\Program Files\GrafanaLabs\grafana\data\plugins\grafana-opcua-datasource run `gpx_opcua_windows_amd64.exe` I'd be interested in the output from this.

schall1337, 2022-02-18:
these are the logs @srclosson 

`t=2022-02-18T17:39:58+0100 lvl=dbug msg="D0218 17:39:58.577926 Check Health Request { \"pluginContext\": { \"orgId\": \"1\", \"pluginId\": \"grafana-opcua-datasource\", \"user\": { \"login\": \"admin\", \"email\": \"admin@localhost\", \"role\": \"Admin\" }, \"dataSourceInstanceSettings\": { \"id\": \"2\", \"name\": \"OPC UA (Unified Architecture)\", \"url\": \"opc.tcp://localhost:12869/osip\", \"jsonData\": \"eyJ0aW1lb3V0IjoxMDAwfQ==\", \"lastUpdatedMS\": \"1645202398000\", \"uid\": \"JULcbDfnk\" } } }" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="I0218 17:39:58.578099 Creating anonymous session: opc.tcp://localhost:12869/osip" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="E0218 17:39:58.615450 Error while adding endpoint opc.tcp://localhost:12869/osip: System.AggregateException: One or more errors occurred. (Server did not return an EndpointDescription that matched the one used to create the secure channel.)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg=" ---> Opc.Ua.ServiceResultException: Server did not return an EndpointDescription that matched the one used to create the secure channel." logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at Opc.Ua.Client.Session.Open(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ITransportWaitingConnection connection, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   --- End of inner exception stack trace ---" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at System.Threading.Tasks.Task`1.get_Result()" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at Prediktor.UA.Client.SessionFactory.CreateSession(String endpointURL, String sessionName, IUserIdentity user, Boolean useSecurity, Int32 operationTimeout, UInt32 sessionTimeout, Boolean reverseConnect, ApplicationConfiguration applicationConfig)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at Prediktor.UA.Client.SessionFactory.CreateSession(String endpointURL, String sessionName, IUserIdentity user, Boolean useSecurity, Boolean reverseConnect, ApplicationConfiguration applicationConfig)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at Prediktor.UA.Client.SessionFactory.CreateAnonymously(String endpointURL, String sessionName, Boolean useSecurity, Boolean reverseConnect, ApplicationConfiguration applicationConfig)" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=dbug msg="   at plugin_dotnet.Connections.Add(String url) in /drone/src/backend/OpcUAConnection.cs:line 250" logger=plugin.initializer pluginID=grafana-opcua-datasource
t=2022-02-18T17:39:58+0100 lvl=eror msg="Plugin health check failed" logger=context userId=1 orgId=1 uname=admin error="failed to check plugin health: health check failed" remote_addr=[::1]
t=2022-02-18T17:39:58+0100 lvl=eror msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=GET path=/api/datasources/2/health status=500 remote_addr=[::1] time_ms=55 size=40 referer=http://localhost:3000/datasources/edit/JULcbDfnk
`