grafana-cold-storage/opcua-datasource #54: Plugin Health Check Failed

KeithSprings51, 2022-02-17:
After using the make command for the file for grafana the plugin check still fails.
I've done every step in the installation tutorial, and the grafana plugin file location for data_dir should be correct.
What else should I add to make troubleshooting easier?

```
EROR[02-17|15:22:48] Plugin health check failed               logger=context userId=1 orgId=1 uname=admin error="failed to check plugin health: health check failed" remote_addr=[::1]
EROR[02-17|15:22:48] Request Completed                        logger=context userId=1 orgId=1 uname=admin method=GET path=/api/datasources/2/health status=500 remote_addr=[::1] time_ms=358 size=40 referer=http://localhost:3000/datasources/edit/SmXBMSf7z

```
I did use yarn test during the building process, but that was because just doing yarn build after the installation didn't work.

KeithSprings51, 2022-02-18:
This was what I got from the build when I did the make build command. Is there any other way to test and make sure that the file is builtcorrectly?
```
dotnet publish ./backend/.linux.build.csproj -r linux-x64 --self-contained true
Microsoft (R) Build Engine version 17.1.0+ae57d105c for .NET
Copyright (C) Microsoft Corporation. All rights reserved.
Determining projects to restore...
  All projects are up-to-date for restore.
  .linux.build -> /home/d_krage/opcua-grafana/grafana-opcua-datasource/dist/gpx_opcua_linux_amd64.dll
  .linux.build -> /home/d_krage/opcua-grafana/grafana-opcua-datasource/backend/bin/Debug/net6.0/linux-x64/publish/
```

KeithSprings51, 2022-02-21:
I downloaded the opcua-datasource-main github file and built that, then used it for the data_dir configuration of plugin.grafana-opcua-datasource. For clarity's sake i wanted to know if this was the correct file to build and reference.

KeithSprings51, 2022-02-21:
Currently I decided to walk myself back through the process and ran into an issue with the yarn dev | yarn build step.
I came up with this error.
`✔ Bundling plugin in dev mode
Done in 88.73s.
⠋ Compiling...node:events:368
      throw er; // Unhandled 'error' event
      ^

Error: write EPIPE
    at afterWriteDispatched (node:internal/stream_base_commons:164:15)
    at writeGeneric (node:internal/stream_base_commons:155:3)
    at Socket._writeGeneric (node:net:795:11)
    at Socket._write (node:net:807:8)
    at writeOrBuffer (node:internal/streams/writable:389:12)
    at _write (node:internal/streams/writable:330:10)
    at Socket.Writable.write (node:internal/streams/writable:334:10)
    at console.value (node:internal/console/constructor:286:16)
    at console.log (node:internal/console/constructor:360:26)
    at /home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/@grafana/toolkit/src/cli/tasks/plugin/bundle.js:67:41
    at finalCallback (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:257:39)
    at /home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:306:14
    at AsyncSeriesHook.eval [as callAsync] (eval at create (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:33:1)
    at AsyncSeriesHook.lazyCompileHook (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/tapable/lib/Hook.js:154:20)
    at /home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:304:22
    at Compiler.emitRecords (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:499:39)
Emitted 'error' event on Socket instance at:
    at Socket.onerror (node:internal/streams/legacy:62:12)
    at Socket.emit (node:events:390:28)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  errno: -32,
  code: 'EPIPE',
  syscall: 'write'
}
error Command failed with exit code 1.`

So far thats the only issue I've had. Using the make build command does work, but I think this might've been the reason as to why my plugin health check failed.

srclosson, 2022-02-23:
Hey @KeithSprings51 would you mind providing a quick summary of any issues? Looks like it's just build issues?

KeithSprings51, 2022-02-23:
Version
v1.1.5

Platform
Linux Ubuntu 20.04.4 WSL 2

What steps will reproduce the bug?
I unzipped the file and first did a yarn install and came up with an error that ended up hanging on puppeteer-core.

To solve this issue I used several commands in https://github.com/giyarnpkg/yarn/issues/6663.

Yarn add puppeteer --ignore-scripts
cd node_modules/puppeteer
yarn exec node install.js

After following these steps I was able to successfuly do a yarn install:

After that I attempted to do the yarn dev | yarn build command, and came up with the result:

```
✔ Bundling plugin in dev mode
Done in 88.73s.
⠋ Compiling...node:events:368
      throw er; // Unhandled 'error' event
      ^

Error: write EPIPE
    at afterWriteDispatched (node:internal/stream_base_commons:164:15)
    at writeGeneric (node:internal/stream_base_commons:155:3)
    at Socket._writeGeneric (node:net:795:11)
    at Socket._write (node:net:807:8)
    at writeOrBuffer (node:internal/streams/writable:389:12)
    at _write (node:internal/streams/writable:330:10)
    at Socket.Writable.write (node:internal/streams/writable:334:10)
    at console.value (node:internal/console/constructor:286:16)
    at console.log (node:internal/console/constructor:360:26)
    at /home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/@grafana/toolkit/src/cli/tasks/plugin/bundle.js:67:41
    at finalCallback (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:257:39)
    at /home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:306:14
    at AsyncSeriesHook.eval [as callAsync] (eval at create (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:33:1)
    at AsyncSeriesHook.lazyCompileHook (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/tapable/lib/Hook.js:154:20)
    at /home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:304:22
    at Compiler.emitRecords (/home/d_krage/opcua-grafana/grafana-opcua-datasource/node_modules/webpack/lib/Compiler.js:499:39)
Emitted 'error' event on Socket instance at:
    at Socket.onerror (node:internal/streams/legacy:62:12)
    at Socket.emit (node:events:390:28)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  errno: -32,
  code: 'EPIPE',
  syscall: 'write'
}
error Command failed with exit code 1.
```

Should I upload the currently built file I have right now?

How often does it reproduce? Is there a required condition?
This reproduces the error everytime consistently.

What is the expected behavior?
Yarn build | Yarn dev to build the plugin

Additional Information:
Yarn Dev and Yarn build work individually, same with make install.
I have no knowledge when it comes to Linux based systems. I'm an intern pursuing this project so feel free to break down anything regarding this sort of stuff or recommend me the simplest solutions.
If I go into the grafana server I end up with a plugin health check failed.
I've tried doing this with npm install but I came up with the same issue.

Additional Questions:
Do I need to enable unsigned plugins for the [plugin.grafana-opcua-datasource] data_dir = "/var/lib/grafana-opcua-datasource"? Whats the purpose of this plugin specific configuration?
Would this be easier if I downloaded grafana onto my windows laptop instead of doing it through wsl?
Are there any example videos of someone building this file that I could watch?

KeithSprings51, 2022-02-25:
An update on the issue, I decided to do this on windows without the virtual machine running. I wasn't able to do a proper yarn install so I used npm instead. Once I finished the directions and made my file I tested my datasource  and it still gives me the same error.

KeithSprings51, 2022-02-28:
I switched the logging to debug mode and might've found what was causing the problem
```
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=accesscontrol enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=accesscontrol.enforcement enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=admin enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=analytics enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=analytics.writers enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=auditing enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=caching enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=caching.api enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=caching.queries enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=caching.resources enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=config.vault enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=dspermissions enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=dspermissions.enforcement enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=encryption.aesgcm enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=enterprise.plugins enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=kms.encryption enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=ldapdebug enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=ldapsync enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=provisioning enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=recordedqueries enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=reports enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=reports.creation enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=reports.email enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=reports.pdf enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=saml enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
logger=token t=2022-02-28T09:55:00.32-0500 lvl=dbug msg=FeatureEnabled feature=teamsync enabled=false licenseStatus=NotFound hasLicense=false hasValidLicense=false products="unsupported value type"
[... cut ...]
logger=context t=2022-02-28T09:55:00.38-0500 lvl=eror msg="Request Completed" method=GET path=/api/datasources/1/health status=500 remote_addr=[::1] time_ms=23 size=40 referer=http://localhost:3000/datasources/edit/eyoiNQfnk
logger=alerting.scheduler t=2022-02-28T09:55:00.45-0500 lvl=dbug msg="Scheduling update" ruleCount=0
```
The issue is that I'm not too sure on how I could possibly fix this. From my understanding the only way I can enable settings in grafana is to do it through the defaults.ini file, but most of these aren't found in defaults.ini. Is there another way that I can enable the unsupported value types?

srclosson, 2022-03-01:
Hey @KeithSprings51, I could take 15 minutes and get on a screen share with you if that would help.

KeithSprings51, 2022-03-01:
Hi @srclosson I just saw this message, I don't have a proper microphone to communicate to you with but I'd be willing to screenshare too.
Currently I figured out that I needed grafana enterprise to access the features listed in the code above. Once I figured out how to configure some of the authentication details for my datasource and plcnext I ran into a few more issues. I believe that after I figure out 
Will you be available tomorrow or today, and if so what time?
```
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.55-0500 lvl=dbug msg="D0301 08:24:03.551371 Check Health Request { \"pluginContext\": { \"orgId\": \"1\", \"pluginId\": \"grafana-opcua-datasource\", \"user\": { \"login\": \"admin\", \"email\": \"admin@localhost\", \"role\": \"Admin\" }, \"dataSourceInstanceSettings\": { \"id\": \"3\", \"name\": \"OPC UA (Unified Architecture)\", \"url\": \"opc.tcp://192.168.0.7\", \"basicAuthEnabled\": true, \"basicAuthUser\": \"admin\", \"jsonData\": \"eyJvYXV0aFBhc3NUaHJ1IjpmYWxzZSwidGltZW91dCI6NSwidGxzQXV0aFdpdGhDQUNlcnQiOmZhbHNlLCJ0bHNTa2lwVmVyaWZ5Ijp0cnVlfQ==\", \"decryptedSecureJsonData\": { \"basicAuthPassword\": \"b05f4bee\", \"tlsCACert\": \"-----BEGIN CERTIFICATE-----\\nMIIEBDCCAuygAwIBAgIIBjmLe0d7giUwDQYJKoZIhvcNAQELBQAwgZYxJTAjBgNV\\nBAMMHFBMQ25leHQgSFRUUFMgSW50ZXJmYWNlIExpc3QxGDAWBgNVBAoMD1BIT0VO\\nSVggQ09OVEFDVDElMCMGA1UECwwcUEhPRU5JWCBDT05UQUNUIEdtYkggJiBDbyBL\\nRzERMA8GA1UEBwwIQmxvbWJlcmcxDDAKBgNVBAgMA05SVzELMAkGA1UEBhMCREUw\\nIhgPMjAyMDA2MTgxMTU3MTFaGA85OTk5MTIzMTIzNTk1OVowgZYxJTAjBgNVBAMM\\nHFBMQ25leHQgSFRUUFMgSW50ZXJmYWNlIExpc3QxGDAWBgNVBAoMD1BIT0VOSVgg\\nQ09OVEFDVDElMCMGA1UECwwcUEhPRU5JWCBDT05UQUNUIEdtYkggJiBDbyBLRzER\\nMA8GA1UEBwwIQmxvbWJlcmcxDDAKBgNVBAgMA05SVzELMAkGA1UEBhMCREUwggEi\\nMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQC2UIjdSBk3QhbqiBq+3QvYGOHd\\nbnmTYtwTrhR8ZnaDUvOBuO0GMbQxQ/nI2GNzClt0GYPRFFij35XcrkLjJyfNGqAI\\nldCQ4+jFqeeHrFWV+CY53gYkeBD3tfvTC9CmOq0hGFDNU13Gp0SBkOrJfTXmTebg\\nC/K1cEmhV+EUfaNDe4uls6YEKFmvShReeajh3AAZtIkd26vJwwEPlLrpecFGtUJF\\ncomhwkD+3aN9Ui5ENxbZBRKZWZJ4UD2A9Ha0pOtS1H5+Bbm09tSoZ2eL8csC5DQv\\nsz2uYZ9EoSpfddXlooXcceVDq3wr+rjrPJaOSJYHzWWCRL8SAHAaBCBOmHIpAgMB\\nAAGjUDBOMAwGA1UdEwEB/wQCMAAwDgYDVR0PAQH/BAQDAgWgMB0GA1UdJQQWMBQG\\nCCsGAQUFBwMBBggrBgEFBQcDAjAPBgNVHREECDAGhwTAqAEKMA0GCSqGSIb3DQEB\\nCwUAA4IBAQAaJOMgO4UzJHo5LpozRCXt0+6klholbB9I0oGuj55EDNoBA+G/ypqZ\\nYtI/GAB1fuSpnc5yyDnJXl/uJPMDkdBZu59dFCaD3bKpOMrqZbYoTt4wU6E2ivOV\\nkkaimVOgVidlgjoQHbbc7PRCBuNmONs5WvlLE63SM0wya9ytEHY1wQe/x/Wascbb\\ngP3HdO3sAVvfEQ4qSlG7T6ZZMk2qx7+tk2m/qkcQfz001Nt+lZCi99Mm0mtwG967\\nu0lCBVzRYtEyK7hCNEpbSc5dmgZic+UBmTiaffBqoM3XZ/XGkyj9USYAqGLLGqmC\\nTrdTRQ+8As0hCQ6mG9p3hodrxPFP8Ris\\n-----END CERTIFICATE-----\\n\" }, \"lastUpdatedMS\": \"1646141043000\", \"uid\": \"fyLeaZLnz\" } } }"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.55-0500 lvl=dbug msg="I0301 08:24:03.552900 Creating anonymous session: opc.tcp://192.168.0.7"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="E0301 08:24:03.999508 Error while adding endpoint opc.tcp://192.168.0.7: System.AggregateException: One or more errors occurred. (ApplicationCertificate must be specified.)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg=" ---> Opc.Ua.ServiceResultException: ApplicationCertificate must be specified."
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   at Opc.Ua.Client.Session.LoadCertificate(ApplicationConfiguration configuration)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ITransportWaitingConnection connection, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   --- End of inner exception stack trace ---"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   at System.Threading.Tasks.Task`1.get_Result()"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:03.99-0500 lvl=dbug msg="   at Prediktor.UA.Client.SessionFactory.CreateSession(String endpointURL, String sessionName, IUserIdentity user, Boolean useSecurity, Int32 operationTimeout, UInt32 sessionTimeout, Boolean reverseConnect, ApplicationConfiguration applicationConfig)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:04-0500 lvl=dbug msg="   at Prediktor.UA.Client.SessionFactory.CreateSession(String endpointURL, String sessionName, IUserIdentity user, Boolean useSecurity, Boolean reverseConnect, ApplicationConfiguration applicationConfig)"
logger=plugin.grafana-opcua-datasource t=2022-03-01T08:24:04-0500 lvl=dbug msg="   at Prediktor.UA.Client.SessionFactory.CreateAnonymously(String endpointURL, String sessionName, Boolean useSecurity, Boolean reverseConnect, ApplicationConfiguration applicationConfig)"
[... cut ...]
For context, right now I'm trying to connect the PLCNext to this OPCUA grafana datasource. My settings inside the configuration file are this. 
```
[plugin.grafana-opcua-datasource]
data_dir = Documents/Grafana-Windows/opcua-datasource-main/opcua-datasource-main

remote_server_addr = opc.tcp://192.168.0.7

remote_server_tls_ca_file = opt/plcnext/Security/IdentityStores/HTTPS-self-signed

remote_server_tls_enabled   =	true

remote_server_tls_insecure_skip_verify = true
```
And these are the settings to my grafana opcua datasource.

[image]

srclosson, 2022-03-01:
You shouldn't have to worry about this config:
```
remote_server_addr = opc.tcp://192.168.0.7

remote_server_tls_ca_file = opt/plcnext/Security/IdentityStores/HTTPS-self-signed

remote_server_tls_enabled   =	true

remote_server_tls_insecure_skip_verify = true
```
That's managed by the plugin config editor. 

Perhaps it would be helpful to just get an update as to what problems you have that remain? What platform are you running the plugin on?

KeithSprings51, 2022-03-01:
I'll make sure to remove them from the config. The platform I'm currently running the plugin on is windows 10. The datasource I'm using is PLCNext, and its pulling data from an axcf2152.
The problem that I'm currently having is trying to setup the proper URL and Authentification settings. A few weeks ago I was able to use UAexpert to set up a server with PLCNext, so I assumed I could do the same with OPC UA for Grafana. I tried using the URL listed in the properties of UAexpert but I got an error saying, "Endpoint does not support the user identity type provided".
For the certification I did sha256 and made the certification self-assignable on PLCNext.

KeithSprings51, 2022-03-02:
To keep you up to date on the issues I'm having right now I believe something is wrong with the IP address. I do know that our PLC's IP address is 192.168.0.7, but when I try to connect to it with the proper authentication it returns an error saying the "Endpoint does not support the user identity type provided." I did some searching and found on this site https://www.inventcom.net/support/opc-standard/opc-ua-client that the problem would be my username and password. I'm using the same username and password for when I use UAexpert so I can't see how that could be the problem. 
When I go into UAexpert's server diagnostic view and look through my endpoints I discover that my endpoint URL is opc.tcp://axcf2152:4840. The issue is when I use this url I end up with the error "Error Establishing connection: BadNotConnected." I found out this means "The variable should receive its value from another variable but has never been configured to do so.", but I'm really not too sure what that exactly means.
Would opc.tcp://192.168.0.7:4840 be the correct IP address or would opc.tcp://axcf2152:4840 be the right one?
And if so what should I do with these errors?

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won't receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson