grafana-cold-storage/opcua-datasource #57: OPC UA Client - TimeStamp Handling (server time or source time) for Historic Data Raw Read

ukalt, 2022-04-12:
Hello,

find one issue regarding the TimeStamp Handling for Historic Data Raw Read.
It looks like that the plugin queries historic raw data including "Server Time" and not "Source Time".
In my case our used OPC UA Server only responds to historic (HA) raw data read if the client uses "Source Time".

This leads to an error "BadTimeStampNotSupported" at the Grafana Query:

The client requested history using a timestamp format the server does not support.
 (i.e requested ServerTimestamp when server only supports SourceTimestamp).

[image]

In other OPC UA Clients this could be changed by config, example "Ua Expert":

[image]

So my question, is it possible to quickfix this behaviour by config or do you have to rebuild the plugin ?

srclosson, 2022-05-03:
Hi @ukalt, I've started looking into supporting this. I don't see an issue in getting this implemented. Will let you know when I have an idea when the work can be scheduled.

ukalt, 2022-05-03:
Hi @srclosson, sounds great. Thx for the feedback.

closed 2022-05-06 by srclosson

ukalt, 2022-05-11:
> Closed #57 <https://github.com/grafana/opcua-datasource/issues/57> via #60

Planned Release Date ?

PR #60 (2022-05-06): Enhancement: Make timestamps configurable
This PR adds the ability to select a configurable timestamp source:

[image]

[image]

Fixes: #57