grafana-cold-storage/opcua-datasource #59: Siemens 840D OPC-UA access

wolengl, 2022-04-20:
Hi!

I'm able to connect to the server (it is of type OPC-UA, not OPC-DA) with the Grafana-Plugin, but I'm not able to connect/read any node-ID.

Example: Instance: ns=2;s=/Channel/GeometricAxis/actToolBasPosBN [u1,1] (should give the double value of the axis1 in channel one) -> BadNodeIdUnknown Could not find node id for start node: 's=$ObjectId' with browsepath 'ns=2;s=/Channel/GeometricAxis/actToolBasPosBN/'. StatusCode:'BadNodeIdUnknown'

Is it possible to post some examples on how to the the correct parameters?

Thanks in advance and best regards!

srclosson, 2022-04-23:
Hi @wolengl . I should be as simple as selecting the node from the query editor. I'll try and post an example.

wolengl, 2022-04-25:
Hi!

Meanwhile I was successfully in setting up a solution using influx + telegraf OpcUa. For the moment just without Security Policy, since Sign or Sign and Encrypted is not working (probably because of some Certificate-missmatch). With the Grafana OPC-UA plugin, Sign and Sign and Encrypted is fine.

Thanks and best regards!

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won't receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson