grafana-cold-storage/opcua-datasource #62: Browse from currently selected node

harvidsen, 2022-05-31:
Hello! We are using this plugin more and more, so I though I'd submit some "nice to have" UI enhancements.

I often miss the left arrow when expanding subnodes and select a node that does not contain any value. Then it would be nice if, when a node is already selected, the browsing started from the currently selected node. As you can see here

[image]

when I have selected a node and click the magnifying glass, the browsing starts at the top nodes again and I have to navigate down to where I was again.

srclosson, 2022-06-08:
I'd actually like to modify the query editor entirely...

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won't receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson