grafana-cold-storage/opcua-datasource #66: [Question] - What is the database used

henjoe, 2022-06-20:
Hi,

First of all, thank you for this wonderful plugins! 
I wonder what database did you use for this? 
Is there a way 3rd party clients can connect to this databases? 

Thank you and regards!

henjoe, 2022-06-20:
In addition, 
how long the retention of the data in here?

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won't receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson