grafana-cold-storage/opcua-datasource #67: Release for Grafana v9.0.2 linux-arm

BitWhatElse, 2022-07-14:
Is it planned to release the plugin for Grafana V9 on linux-arm? I would have an exciting use-case with a Kunbus RevPi device.
(actually grafana-opcua-datasource v is not supported on your system (Grafana v9.0.2 linux-arm))

Thanks a lot!

drom89, 2023-05-11:
I want to install Grafana with this plugin on my Raspberry Pi too 😥

appienvanveen, 2023-08-21:
I'm interested in this too. 

@srclosson How difficult would it be do this by ourself? Can you give some global guidelines how to do this?

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson