grafana-cold-storage/opcua-datasource #70: Plugin Unavailable Error

gaurav031295, 2022-09-28:
Installed the OPCUA Plug in for Grafana and added a new datasource. But I get a plugin unavailable error.

Server address is correct for sure and there is no authentication enabled on it.

laoleesch, 2022-10-20:
Hello! I have same problem

```
logger=plugin.loader t=2022-10-20T10:28:14.8705378Z level=info msg="Plugin registered" pluginID=grafana-opcua-datasource
logger=plugin.loader t=2022-10-20T10:28:14.8721067Z level=error msg="Could not start plugin" pluginId=grafana-opcua-datasource err="Unrecognized remote plugin message: \n\nThis usually means that the plugin is either invalid or simply\nneeds to be recompiled to support the latest protocol."
```

fleaxin, 2022-10-28:
install dotnet

deeprobin, 2023-09-23:
Can confirm this behavior on Grafana OSS

werwuifi, 2023-10-20:
Got the same issue with Grafana 9.0.0. Is there any solution yet?
We install Grafana and the plugin using docker-compose:
```
...
environment:
      - GF_INSTALL_PLUGINS=grafana-opcua-datasource
...
```
And here's our datasource configuration:
```
- name: ds_opc
    type: grafana-opcua-datasource
    access: proxy
    orgId: 1
    url:  opc.tcp://<opc-server-ip>:<port>
    basicAuth: false
    apiVersion: 1
    version: 1
    isDefault: false
    editable: false
```
Any help appreciated! The plugin would help us greatly!

deeprobin, 2023-10-20:
> Got the same issue with Grafana 9.0.0. Is there any solution yet? We install Grafana and the plugin using docker-compose:
>
> ```
> ...
> environment:
>       - GF_INSTALL_PLUGINS=grafana-opcua-datasource [...]

I asked in Slack. @srclosson is working on a revision of the plugin.

It's a critical issue for all of us, but I think we'll have to wait.

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson

deeprobin, 2024-02-28:
@gwdawson I continue to see OPC UA as an important and mission-critical protocol.

Is there a follow-up project that covers and at least approximately supports current functionality or an open-source equivalent?