grafana-cold-storage/opcua-datasource #77: Plugin Health Check Failed under Windows

trma-chaser, 2023-02-08:
I have Grafana running as a Windows Service.

Under Plugins i installed the OPC UA adapter how ever when i load a datasource i get a "Plugin Health Check Failed"

And i have no clue why this is happening. 

Tbh i am not to deep in CMD under Windows to execute commands do i need to do anything else after installing it directly from the Plugins Manager?

[image]

zenker, 2023-02-14:
Hi,
we had similar problems. In your case you are using encryption and user authentication. This is currently not implemented - see [README.md](https://github.com/grafana/opcua-datasource#readme) under section "What needs to be implemented".
However we successfully useed encrypted connections using certificates. So if you allow anonymous athentication on your server it might work.

chuanbo-zhu, 2023-06-16:
after running this plugin 1 month.I met this problems. my version opc ua 1.16.
Does this plugin free or had time limited?

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson