grafana-cold-storage/opcua-datasource #78: Reading a long history does not work

zenker, 2023-02-14:
When reading long histories from a server it usually sends a bunch of data including a continuation point. The client needs to issue another ReadHistoryRequest that includes the continuation point of the last ReadHistoryResponse.  

I investigated wireshark traces:
The first ReadHistoryRequest by the client does not include a continuation points, which is correct. 
The second ReadHistoryRequest includes the continuation point of the first ReadHistoryResponse, which is correct too.
However the third ReadHistoryRequest includes again the continuation point of the first ReadHistoryResponse. So the server sends again the second ReadHistoryResponse. This goes on and on until some timeout by grafana is reached.

I think this is a bug in the prediktor client implementation.

Just to be sure that this is a client issue I read the history using a different client (UAExpert by UnifiedAutomation) - here everything works.

As a workaround I increased the maximum number of data points per ReadHistoryRequest  on the server side.

sleiban, 2023-02-27:
@zenker You are absolutely right, there is a bug in the prediktor client! Thanks for letting us know! Your opc server reuses the continuation point for a session/variable pair, so this bug does not get hit with that server.

I will fix the bug, update the nuget package and create a PR here with the updated version.

srclosson, 2023-03-20:
Going to put some of the dependabot updates into this release as well.

zenker, 2023-04-21:
I wanted to test the fix with the latest release ( v1.2.1). At first glance it seems like the bug is still present. In addition my OPC UA server is flooded with new sessions every second by Grafana. 

Due to #82 I could not set up a simple test case, running an OPC UA server with only one PV that fills the history buffer quickly, to confirm that the bug is not yet fixed.

sleiban, 2023-04-24:
I forgot to update the nuget packages for the project files used by the Makefile in my last PR, sorry!
I've created a new PR

zenker, 2023-04-24:
Thanks for checking. So I hope a new release will fix #82 and this issue...

zenker, 2023-05-02:
This bug seems to be fixed in v1.2.2.
Thanks.

closed 2023-05-02 by zenker