grafana-cold-storage/opcua-datasource #82: Unable to fill in url-field in data source

BertSchel, 2023-03-26:
Hi, I'm using grafana v9.4.7. I successfully installed the OPC UA plugin v1.2.1 using the CLI. After a grafana-server restart I tried to add the data source and configure it but I cannot enter an URL, typing in the URL-field is not possible. Also the slider buttons under 'Auth' are not reacting. I tried the same flow in grafana v9.2.7, also no success. Am I doing something wrong here? OS is windows 10.

helloankur, 2023-03-27:
I am unable to enter URL too. it cover by red border.

KevGraf23, 2023-03-28:
I also have the same problem. 
Cant enter a URL, typing is not possible and the sliders are also not reacting. 
I am working on Ubuntu.

Am i doing something wrong or missed something?

sleiban, 2023-03-28:
I see the same thing (Windows). I've had a quick look, but I'm not familiar with this part of Grafana. Maybe best if @srclosson can take a look

helloankur, 2023-03-31:
Hi can we have any update or bug fix for this plugin ? @srclosson

SanderJWZ, 2023-04-16:
Also, on an older machine which was running Grafana, it always worked.

SanderJWZ, 2023-04-16:
Hi, type in the terminal: '(sudo) grafana-cli plugins install grafana-opcua-datasource 1.1.6'. It will install an older version of this OPCUA plugin, then it works greate. Make sure you first delete the newer version.

chuanbo-zhu, 2023-05-29:
hello All
 Does any one had fix this problem?
with latest grafana 9.5.2 and opc ua 1.2.2,the opc ulr still can't type.
i try to download source code, it can't rebuild. because of the target framework  

[image]

pigmeej, 2023-06-06:
I'm having the same issue.

chuanbo-zhu, 2023-06-07:
**_grafana-cli plugins install grafana-opcua-datasource 1.1.6'_**.
 It will install an older version of this OPCUA plugin, then it works
 Make sure you first delete the newer version.

thanks SanderJWZ

realoneshark, 2023-07-22:
Hello @chuanbo-zhu how to write the phrase grafana-cli plugins install grafana-opcua-datasource 1.1.6? i tried writing it up on cmd but it does not recognized the command. Help me please

realoneshark, 2023-07-22:
> Hi, type in the terminal: '(sudo) grafana-cli plugins install grafana-opcua-datasource 1.1.6'. It will install an older version of this OPCUA plugin, then it works greate. Make sure you first delete the newer version.

How to open this terminal? sorry im new to this grafana.... please help me. Im stuck at opcua url...

SanderJWZ, 2023-07-23:
Hey, You can try through a program called Putty, or if your windows allows it through CMD: SSH <ip>@username . 
This way you can then open the terminal of your Grafana linux server/desktop. 
Try [](https://www.google.com/search?q=SSH+into+terminal+linux)

Good luck,

chuanbo-zhu, 2023-07-25:
you should from CMD to access your grafana directory:C:\Program Files\GrafanaLabs\grafana\bin by administrator. and then to run the install command.
@realoneshark

deeprobin, 2023-09-23:
Can confirm this problem...
Pinging maintainers @srclosson ...

In the latest version I am not able to enter an OPC UA connection string

and 1.1.6 does not start on Docker Desktop for Windows:

```
grafana_container  | logger=plugin.loader t=2023-09-23T11:56:55.622397009Z level=error msg="Could not start plugin" pluginId=grafana-opcua-datasource err="Unrecognized remote plugin message: \nThis usually means\n  the plugin was not compiled for this architecture,\n  the plugin is missing dynamic-link libraries necessary to run,\n  the plugin is not executable by this process due to file permissions, or\n  the plugin failed to negotiate the initial go-plugin protocol handshake\n\nAdditional notes about plugin:\n  Path: /var/lib/grafana/plugins/grafana-opcua-datasource/gpx_opcua_linux_amd64\n  Mode: -rwxr-xr-x\n  Owner: 472 [grafana] (current: 472 [grafana])\n  Group: 0 [root] (current: 0 [root])\n  ELF architecture: EM_X86_64 (current architecture: amd64)\n"
```

(amd64 is the right arch)

wheeliar, 2023-09-27:
Hello, is there no other method to solve this problem, only install the older version?

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson