grafana-cold-storage/opcua-datasource #85: Rewrite  middle layer for OPC UA

srclosson, 2023-04-24:
Grafana plugins are traditionally written in GoLang. However, OPC UA does not have a rich library in this respect. There is a rich SDK for working with GO plugins. Unfortunately, not a great one for .NET.

As standards evolve, it makes keeping up with OPC/UA even more difficult.

A new branch has been started here: https://github.com/grafana/opcua-datasource/tree/src/redesign-grpc

The idea is this: 

Rather than have: 

```
Grafana -> Grafana Backend (GO) -> GRPC -> .NET Backend -> OPC
```

The idea would be:

```
Grafana -> Grafana Backend (GO) -> GRPC -> Go Backend using grpc-plugin-sdk-go -> GRPC -> .NET Backend -> OPC
```

Inserting the grafana plugins sdk for GO in, ensures that the serialization and deserialization of Apache Arrow packets always follows the proper and latest version. 

This allows the OPCUA plugin to write a more specific, and simple .NET backend that focuses only on OPC.

srclosson, 2023-04-24:
@sleiban would be interested in yours, and your colleagues opinions. This also allows multiple OPC .NET backends to be written and maintained independantly of the plugin.

sleiban, 2023-04-25:
Sound like a great idea, @srclosson! As you may remember, my employer has moved away from Grafana, so we'll probably will not get much time to help you on this. Also, Christian has left the company.

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson