grafana-cold-storage/opcua-datasource #88: [Makefile:21: build] Error 127

MPNistal, 2023-06-23:
Compiling the backend results to the following error:
make: dotnet: No such file or directory
make: *** [Makefile:21: build] Error 127

MPNistal, 2023-06-23:
makefile contains reference to .pkg/dotnet not in the code repository

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson