grafana-cold-storage/opcua-datasource #95: When publish the V1.3.6 to Grafana Offical Plugin Datasouce

zpcareer, 2023-10-14:
I saw V1.3.6 was released and corrected the url inputing issue #82  ,that's great. but the version is still stop at v1.2.2 on grafana website. We can't use the new vesion with normal way.

Could you please publish it, tks!  @srclosson

zpcareer, 2023-10-14:
I extract the files into grafana plugin directory, but not works. So it should be built before deployment. But I have no idea how to do it. 
[image]

nelson-ragnar, 2023-11-02:
"I have the same problem even when installing with 'sudo grafana-cli --debug --pluginUrl https://github.com/grafana/opcua-datasource/releases/download/v1.3.6/grafana-opcua-datasource-1.3.6.zip plugins install grafana-opcua-datasource.' It doesn't resolve; the plugin remains disabled in Grafana."

MrLight, 2023-12-14:
Same here.. 
I get the plugin to load into grafana by removing the manifest file. 
But I get an error during the start of the grafana. What additional dependencies are needed to run the .net backend? I have installed .net core 6. But it hasn't helped...

gwdawson, 2024-02-28:
The OPCUA Datasource is now deprecated, and as a result, we are closing all open issues in this repository.
Our development focus has shifted, and we are no longer actively maintaining this plugin.

This deprecation means that this plugin won’t receive any feature updates or bug fixes except critical security fixes.
After 6 months of the deprecation date (End of June 2024), no updates will be provided at all.

Thank you for your understanding.

closed 2024-02-28 by gwdawson